LaserWeb/deprecated-LaserWeb2 #1: Grbl Support: need help

openhardwarecoza, 2016-03-21:
Need help from a grbl owner with a bit of JS experience to add some more if statements below
https://github.com/openhardwarecoza/LaserWeb2/blob/gh-pages/js/spjs.js#L197

you can get sample code from https://github.com/openhardwarecoza/LaserWeb/blob/master/i/main.js#L2011 - I just dont have a grbl machine at hand to test with :)

Cinezaster, 2016-04-05:
Just hooked it up to GRBL (0.9g). 
- Version detection will also be necessary.
- A $X unlock button will be necessary somehow.
- I'll see some grbl code in spjs.js on line 247

Will see if I get some time to make a dump of the communication, and start from there.

openhardwarecoza, 2016-04-05:
Yeah this time round i neglected machine control a little in favour of the cam side (;

I got a Uno and CNC Shield in the mail just today (: so now I can play a bit too. The L-Cheapo 3.5w diode should arrive on Friday!  

I'm going to play with Grbl first - I owe it to all the guys in round 1!  They've been waiting long enough
But as always I appreciate any help!

I have already hooked it the new kit, and installed Grbl 0.9J - commited some fixes already but not pushed:
Pushing now

openhardwarecoza, 2016-04-05:
My new toy :)  [image]

Cinezaster, 2016-04-06:
Lots of garbage messages are comming through. got a pull request f4f6a8bebaa51c570258ead0998ec758c2f767cc that checks for garbage messages. It looks like it works

openhardwarecoza, 2016-04-06:
I gave you push access. I'll be on the road for the rest of this week and next :(

openhardwarecoza, 2016-04-06:
_ Cinezaster commented 13 minutes ago:
"We should know somewhere globally what type of device is connected. There will be more exceptions on the way. Also the onWsMessage function is evaluating to much strings. this could be avoided if we know what device is connected and switch to that proprietary protocol to handle the messages.

After the smoothie or repetier code is executed we don't need to evaluate for Grbl or Repetier.
Refactoring data.D handling in a module or dedicated function might be necessary when we got this one working. "
_

I'm all for fixing that, most of that code comes from copying https://github.com/chilipeppr/widget-spjs/blob/master/widget.js#L1234 and removing the non-cp stuff. Including some things like queue status which I want to bring back...

Cinezaster, 2016-04-06:
Implemented a grbl module that parses all grbl messages. 
Cleaned a couple of lines on the smoothie and repetier parsing. 
Still have to test a few things. Is there gonna be any automated testing involved in the future?

openhardwarecoza, 2016-04-06:
(; i dont have the skillset to add automated tested :(

openhardwarecoza, 2016-04-07:
Looks great!  Once this module is done it will be  a good template to move smoothie/marlin/etc into their own respective modules too!

Cinezaster, 2016-04-07:
Yep, that would be the way to go. But first we need to define the functions that go along with the boards, that way you can treat it as middleware or a plugin.

openhardwarecoza, 2016-04-07:
Yip i'll sit tight till you tell me to.  Modules are greek to me so
anxiously analysing each commit to learn from it (:

openhardwarecoza, 2016-04-07:
@Cinezaster  I am heading home tomorrow. Which version of Grbl are you using (would be good if we use the same for now) 
Stock grbl, or one of the laser specific forks like jtechphotonics?

Cinezaster, 2016-04-07:
GRBL (0.9g)
But I might have changed something in de "config.h"
Have to check that on another computer, where I did the compile and testing previously. 
It is outputting laser status, so it's definitely not standard.

Cinezaster, 2016-04-07:
I'm using [mr beam's version of grbl ](https://github.com/mrbeam/grbl). 
But I'll test the latest version of grbl tomorrow.

openhardwarecoza, 2016-04-08:
Hooked up the L-Cheapo 3.5w http://robots-everywhere.com/re_site/purchase/l-cheapo/ that was sponsored to the project by @Spiritplumber (pics here https://plus.google.com/+PetervanderWalt/posts/96WDKaLtoBa ) to the CNC Shield / Uno and loaded up the MrBeam Grbl

Well done on the work @Cinezaster it connected beautifully.   I did have a bit of trouble in that  https://github.com/openhardwarecoza/LaserWeb2/blob/gh-pages/js/grbl.js#L118 strips out the $??? so when typing it manually i didnt know what to type. I assume you are working towards a configuration widget?   I managed to configure Grbl by changing

```
//var description = data.replace(/.*\(|\)/gi,'').slice(0, -1);
var description = data;
```

I had to disable homing, otherwise it went into Alarm mode on the first move. 

Then, I discovered I forgot to add the laseron and laseroff into the Raster so I added that https://github.com/openhardwarecoza/LaserWeb2/commit/aa6f9f2e95ed8890b4495b18e8e3a5d9ec1bed13

Note: There was a change for windows users https://github.com/openhardwarecoza/LaserWeb2/commit/5cc74801c64957a875e57b945e6422df800d8a7e for #23 - just make sure this didnt break anything for you on Linux (I only have a windows desktop for the next few days)

I had to test a Smoothieboard for #23 , and noted that the DRO was updating as "undefined" so I also did https://github.com/openhardwarecoza/LaserWeb2/commit/756924a37c5bdc66ccc6ea34a9ce6a4c3ed0ac03#diff-0017b423cc06479ab911365135d0af43R386 - I know its not the right way though since it shouldnt be doing undefined vars into the function in the first place.  If you need a Smoothie to test on let me know, I'll gladly send you one for free (;

openhardwarecoza, 2016-04-08:
Oh, forgot the highlight of why I wanted to comment!  But success, I did my first burns!

Had to hook laser PWM to Z Endstop pin on the CNC Shield - its the old version prior to grbl pwm support, so D11 was on the endstop header, no biggie

openhardwarecoza, 2016-04-08:
Tagging in the people from LaserWeb1:  There we guys with suitable hardware who might be helpful to test on LaserWeb2 going forward (; 

@ellisgl @Dikabro @karel314 @athulsnair @bulburDE  @Mad-Onion

Cinezaster, 2016-04-08:
Congrats.

I know, they changed the pinout some grbl version ago, I had the same problem.

About undefined xyz https://github.com/openhardwarecoza/LaserWeb2/commit/756924a37c5bdc66ccc6ea34a9ce6a4c3ed0ac03#diff-0017b423cc06479ab911365135d0af43R386
setBullseyePosition shouldn't be called by an incomplete message, I think?
We should build in some message validation for smoothie as I did for Grbl. As a message is not complete we should consider it corrupt and prevent it from going through? I already implemented this in grbl.js. 
In my case I refreshed my browser without disconnecting an connected again, I received tons of garbage messages, but didn't realized it was because of spjs

Having a 3d printer running on marlin, my mill on tinyG, my laserengraver on Grbl, a 360 scara arm running on a custom made serial buffer over bluetooth and a nodejs generator , I don't own enough machines that needs a controller. The Scara bot is going to be replaced by a arduino Duo running on TinyG2, might as well try the smoothie, as I heard lots of good things about it. I would gladly accept your offer to try out the smoothie if you have

I'll make a smoothie module like grbl, that interpreted messages and validates them. Will do the same for TinyG.

openhardwarecoza, 2016-04-08:
> About undefined xyz 756924a#diff-0017b423cc06479ab911365135d0af43R386

Pm me your postal address (:  peter.plaaswerf@gmail.com

openhardwarecoza, 2016-04-08:
"We should build in some message validation for smoothie as I did for Grbl. " - yip i know I'm jumping the gun a bit.  (: - we'll get there. 

PS, sorry about my messy code, i am always happy to learn from my mistakes, so feel free to comment in line notes on any commits that bother you (:

Cinezaster, 2016-04-08:
@openhardwarecoza 
> I had to disable homing, otherwise it went into Alarm mode on the first move. 
Could be due to limit-switches who are connected wrong. Had the same a while ago. [You could invert homing $23=1 (homing dir invert mask:00000001)](https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#23---homing-dir-invert-mask-intbinary)

openhardwarecoza, 2016-04-08:
In this case i am running the diode on my pick and place and dont have
endstops on it yet (:

athulsnair, 2016-04-11:
@openhardwarecoza 
I changed my control board from GRBL to Smoothie. Haven't finished configuraion yet.

Facing problems with PWM output. Smoothie has the PWM output from 0-3.3V and my laser PSU need 0-5V. Any idea about it.

I'll test as soon as I set up the machine  :)

openhardwarecoza, 2016-04-11:
Checkout the smoothk40 wiki entry ( http://smoothieware.org/blue-box-guide
) on the Smoothieware wiki for a schematic of the wiring with a level
shifter

athulsnair, 2016-04-11:
@openhardwarecoza 
Will this work
http://www.amazon.in/RDL-V1-0-Bi-Directional-Logic-Converter/dp/B00S7MQ5HU?tag=googinhydr18418-21&tag=googinkenshoo-21&ascsubtag=3fc8d7d7-44d9-4dbb-90c0-5ac4d2adc61b

openhardwarecoza, 2016-04-11:
It should

Cinezaster, 2016-04-12:
Refactored the grbl.js module #26

karel314, 2016-04-15:
I'll setup my RPI with Laserweb2 this weekend and will do some testing.

Cinezaster, 2016-04-15:
@karel314 You don't need a RPI for Laserweb2. 

1. You just connect your arduino with grbl to your computer. 
2. Run "Serial port json server"
3. Open your browser go to [http://openhardwarecoza.github.io/LaserWeb2/](http://openhardwarecoza.github.io/LaserWeb2/) 
4. select your serial port and the grbl buffer and hit connect

closed 2016-09-27 by openhardwarecoza