LaserWeb/deprecated-LaserWeb2 #3: Esp8266 (mehrdad)

openhardwarecoza, 2016-03-22:
@mehrdad  so lets talk (: 

I can compile lw2 down to a single file.  We are also working on Fabrica,  see arthur wolf's posts on google plus (follow me on google plus too).  

That parts easy.  Hosting the file from an esp,  probably possible but it could be a few mb so we'll have to see.  

The first thing we need is a websocket server to serial vridge.  You can't telnet from the browser and http requests have too much overhead.  

Once we have that,  i can start porting the code from laserweb1 since its node.js app acted as websocket to serial bridge.   

At the end of the day,  dont worry too much about it.  You can already play with laserweb2 live on http://openhardwarecoza.github.io/LaserWeb2

openhardwarecoza, 2016-03-22:
On a social note,  you are literally on a spam spree.  Just look at yoir public activity.  Opening same issue about thc on literally 10+ projects?   Opening an issue on a comoeting products github and ask them if they saw laserweb yet.  C'mon man,  get responsible please (:

mkeyno, 2016-03-22:
wow, Peter I'm so novice here and I didn't know that ,but thanks for noticing , actually I'm trying to find the piece of puzzle  for this project and probably gather the people for the same issue , actually I dig in github to find the way to create public project and invite the  talented people to join the forum , that's why I asking the people to visit the code like your laserweb and Luc' codes
any way truly it is honest mistake and hope guys like you not offended , but every day I log on githut check any change of codes I'm interested too
again I'm so sorry Peter

openhardwarecoza, 2016-03-22:
No problem,  just keep in mind some of us have been round the block a few
times and know the who's who to call on for help,  and who rather to leave
alone.  Feel free to talk here,  i dont have a issue with healthy
discussion like other projects do (;

openhardwarecoza, 2016-04-07:
Closing this one for now. If you ever want to come back and talk about this, just reopen it (:

closed 2016-04-07 by openhardwarecoza

mkeyno, 2016-04-07:
thanks anyway Peter