LaserWeb/deprecated-LaserWeb2 #7: Position 0,0 top left?

cojarbi, 2016-03-24:
[image]

cojarbi, 2016-03-24:
Location seems to be off once imported no matter where is located in the file ( evenif  illustrator and inkscape location is 0,0 )

Generating code for svg also shows preview off from the imported file.

First laser ON seems lower left of square, that seems to match the path preview generated by LW
G1 F3200 X92.465 Y155.096 S255
[image]
[svg gcode off.txt](https://github.com/openhardwarecoza/LaserWeb2/files/187219/svg.gcode.off.txt)

openhardwarecoza, 2016-03-24:
Already an open issue (:

I havent coded in position at all.  The idea is to look at a bounding box
of the imported three.js object,  see how far away from 0,0 is is.  Then
translate the object.  Store that initial translation values in an array
and then account for it when scaling/moving object.

openhardwarecoza, 2016-03-24:
Duplicate of #4

But I also pushed some code this morning that fixes it for DXFs.  Just note the position boxes at the bottom right:  After placing it on 0,0 they already contain real world values . I'll improve that a bit later

0,0 for SVG works for the inital object but generating gcode places the gcode in the wrong space (svg has 0,0 as top left, everything else has 0,0 bottom left - so a lot of extra code to flip the offset etc needed

closed 2016-03-24 by openhardwarecoza

cojarbi, 2016-03-24:
Woops. Sorry by know you.would think that I'm reading previous issues? I am :)

Saludos / Regards

Ariel Yahni