LaserWeb/deprecated-LaserWeb2 #8: Origin as configurable (for the k40 folks where top left cuts best)

openhardwarecoza, 2016-03-25:
@Gustavvr wanted to help out so creating issue to track this
Notes:

Add a configurable field to https://github.com/openhardwarecoza/LaserWeb2/blob/gh-pages/index.html#L560 (handle the saving and loading to Localsettings here too https://github.com/openhardwarecoza/LaserWeb2/blob/gh-pages/js/main.js#L193

 translation functions: https://github.com/openhardwarecoza/LaserWeb2/blob/gh-pages/js/fileprep.js#L77  - here it puts it at 0,0 - but if we implement the Origin change we need to do that calculation here...  

The way the gcode generator works in LW2 is that we create and move the threejs objects to where we want it. When I run the gcode generator over the threejs geometry it does a localtoWorld to get the coordinates in the scene for each vertex (: (so less math)

Gustavvr, 2016-04-02:
@openhardwarecoza I have submitted a pull request to add this above.  I will add the "right Top|Bottom" quadrants and center when I have some more time later this week.

openhardwarecoza, 2016-04-04:
Pull merged! Thanks Gustav!

Gustavvr, 2016-04-04:
Let's Leave this open untill I add:
**TopRight
BottomRight
Center
TopCenter
BottomCenter**
I don't necessarily need them but it seems like options that should exist.

openhardwarecoza, 2016-04-04:
(: thanks,  will keep open

openhardwarecoza, 2016-05-02:
(; Closing this one for now.  Don't seem to get it as a request anymore from the community

closed 2016-05-02 by openhardwarecoza