LaserWeb/deprecated-LaserWeb2 #9: Tool tips = feature enhancement

funinthefalls, 2016-03-25:
Found this on the web:
http://projects.nickstakenburg.com/tipped/documentation

If you implement this in the overall design, I can go through and add context sensitive tool tips where they make sense, makes for a much more user friendly app in my opinion.

openhardwarecoza, 2016-03-26:
We already have tooltips support in jqueryui (http://api.jqueryui.com/tooltip/) which is anyway there for the sliders and a few other elements already. So wont need t use Nick's 

I've pushed a commit to initialise tooltips in main.js for the entire DOM

That means ANY element that you think needs a tooltip just has to get a title="tooltip here!"

For example, say we have the file open button?

`<span class="btn btn-default btn-file"><i class="fa fa-folder-open fa-fw"></i>Open<input id="file" type="file" accept=".png,.jpg,.jpeg,.bmp,.gcode,.g,.svg,.dxf,.stl" /></span>`

Lets add a toolip to the btn

[<span class="btn btn-default btn-file" ** title="Open a file"**><i class="fa fa-folder-open fa-fw"></i>Open<input id="file" type="file" accept=".png,.jpg,.jpeg,.bmp,.gcode,.g,.svg,.dxf,.stl" /></span>](url)

funinthefalls, 2016-03-26:
Thanks Peter, I will start adding them where I think they will help. Didn't realize that jqueryui  had the ability for tool tips. Learn something new everyday working with you. Slowly my coding skills are coming back.

closed 2016-03-26 by funinthefalls