LaserWeb/deprecated-LaserWeb2 #10: Error loading .svg file

funinthefalls, 2016-03-26:
When I try and load an svg file I get the following error:

TypeError: Cannot read property 'position' of undefined
(http://openhardwarecoza.github.io/LaserWeb2/js/fileprep.js on line 72)

openhardwarecoza, 2016-03-26:
Convert to path first (:

And the svg should only contain paths,  nothing else

funinthefalls, 2016-03-26:
That was the issue, I had one element that was not a path.

closed 2016-03-26 by funinthefalls

openhardwarecoza, 2016-03-26:
Would be a good idea to code some error handling in to alert user to the
fact (;

I'll see what i can do

cojarbi, 2016-03-26:
Could something like this be added  maybe? 

https://github.com/wout/svg.topath.js

AY

openhardwarecoza, 2016-03-26:
That particular one,  sadly no,  since its for svg.js.   We use snap.svg (;
-  try hunting down a plugin for snap.svg

cojarbi, 2016-03-26:
Maybe this https://github.com/jwdavistx/svg-path-finder