LaserWeb/deprecated-LaserWeb2 #15: SPJS server binding to only IPv6

Gustavvr, 2016-04-04:
@openhardwarecoza I just want to know if you have any thoughts on this issues before I start making changes.  

The Go server portion of SPJS is binding to only IPv6 on my computer ( I use IPv6 so I don't want to disable it)  The spjs.js file in LaserWeb2 does not support IPv6 address format ( e.g. [::1] ).  I can look into resolving this.  But I wanted your input before making changes.

openhardwarecoza, 2016-04-04:
You are welcome to make a change on LaserWeb and send a pull request to
support ipv6.

Gustavvr, 2016-04-04:
This issue is on the SPJS server side.  LaserWeb2 works with Ipv6 using brackets e.g. [::1] 
I will try to address this in the server side.  PM me if you need a quick and dirty work around.

closed 2016-04-04 by Gustavvr

openhardwarecoza, 2016-04-04:
Pick it with @johnlauer on the chilipeppr/serial-port-json-server
repository?  (;

johnlauer, 2016-04-04:
Totally open to it.