LaserWeb/deprecated-LaserWeb2 #28: If Tool Offset: 0, position is ignored for SVG generate G-code

Cinezaster, 2016-04-15:
I started with a big svg, scalled it, set tool offset to 0, then changed position. 
Hit generate G-code. position is ignored
Change tool offset to 0.1 or more or less then hit generate G-code. position is applied.

openhardwarecoza, 2016-04-15:
I know about this one, but havent been able to find a fix yet.  

In the generate gcode code:

https://github.com/openhardwarecoza/LaserWeb2/blob/gh-pages/js/threegcode.js#L66

We check is an inflated path present, then generate gcode from that
If its sliced stl, if its a dxf and finally if its a SVG - this is just so we can grab the correct parameters and object type to iterate through to generate gcode from.

The catch comes in in that svg has reversed coordinates in Y (top left = 0,0,   for everything else bottom left is 0,0 so we need to flip it while loading it https://github.com/openhardwarecoza/LaserWeb2/blob/gh-pages/js/svg-render.js#L170

and then come time to generate gcode we also need to keep this reversed coordinates in mind:  
Which I try doing here https://github.com/openhardwarecoza/LaserWeb2/blob/gh-pages/js/threegcode.js#L187 - but as you can see - thats putting Y coordinates in weird places.

I havent figured out why though.  The difference between the svg's position and the gcode's position doesnt match the laserymax /2, or the bbox on the svg.... The difference between the two also differs per file.

openhardwarecoza, 2016-04-16:
What is interesting,  the inflate path code,  and generate gcode both iterate through the children of the fileParentGroup and work off the localToWorld points in the children.    So why is inflatepath working and generate gcode isnt.  (note nb i do have a function at the bottom of fileprep.js that repositions the inflatepath afterwards.  Could be possible to reuse the math.  On my way to airport today for return flight so no time to dig in :(

openhardwarecoza, 2016-05-09:
Right, svgreader.js in place - closing this issue for now.

I'll be adding some enhancements for SVG soon again (per color power/speed etc) but for now what I have in is a direct replacement for what we had with snap.svg

closed 2016-05-09 by openhardwarecoza