LaserWeb/deprecated-LaserWeb2 #29: Cannot read property 'position' of undefined

cojarbi, 2016-04-19:
I get this when loading an svg or inkscake svg
Same file in LW1 works ok.

[LW2 copy.svg.txt](https://github.com/openhardwarecoza/LaserWeb2/files/225226/LW2.copy.svg.txt)
[LW2P copy.svg.txt](https://github.com/openhardwarecoza/LaserWeb2/files/225227/LW2P.copy.svg.txt)

TypeError: Cannot read property 'position' of undefined
(http://openhardwarecoza.github.io/LaserWeb2/js/fileprep.js on line 72)

openhardwarecoza, 2016-04-19:
Only paths.  Convert everything to paths,  and remove non path objects.
Thats becoming a faq (;

This is a cnc after all (;

cojarbi, 2016-04-19:
I'm going to hit the computer with a hammer. Inkscape says they all are paths with N amount of nodes

Cinezaster, 2016-04-19:
File is fine, contains only paths.
The paths are wrapped by two `<g> <g>...paths </g> </g>` removed one and then it works.

cojarbi, 2016-04-19:
@Cinezaster good to know that I'm not mad. How can we prevent that from becoming a problem or why is Inkscape doing that?

ellisgl, 2016-04-19:
It looks like a known issue: http://wiki.inkscape.org/wiki/index.php/Save_Cleaned_SVG

Cinezaster, 2016-04-19:
I only know how you can prevent this in illustrator. 
We should find a fix for this in the code, i'm already looking at it. 
Problem is that different groups can have different translations which causes problems (in laserweb).
Save_Cleaned_SVG is the best option for now.

cojarbi, 2016-04-19:
How do you prevent it with illustrator? It's great if you can fix it on the code, that way you prevent futures cases like mine or and thus giving a great experience to the user

ellisgl, 2016-04-19:
In inkscape - you can save as a plain SVG (second item on the drop list when saving as). See if that helps. I've done a couple tests, but couldn't get the double g's.

openhardwarecoza, 2016-04-19:
From what I understand, we can run the svg through flatten.svg (https://gist.github.com/timo22345/9413158) to fix the transforms... Tricky one

Cinezaster, 2016-04-19:
In illustrator select a group and then degroup. Do this for all groups!

openhardwarecoza, 2016-04-19:
Ok i tried out flatten.js but still have a <g> inside a <g>

openhardwarecoza, 2016-04-19:
After this commit:  
https://github.com/openhardwarecoza/LaserWeb2/commit/2e2c6d23bf2c6236e6d8192616ae5bea12523aaf

We now get this...

[image]

openhardwarecoza, 2016-04-19:
Okay definately breaks here: on line 124  https://github.com/openhardwarecoza/LaserWeb2/blob/gh-pages/js/svg-render.js#L124

(Uncommenting breaks it again):

`//path = path.transform(path.parent().transform().global);`

path.parent:?  Which parent. It has no parent (;

[image]

openhardwarecoza, 2016-04-19:
Further digging:  Paths that work have G as a parent. Paths that done have a M as a parent

openhardwarecoza, 2016-04-19:
I can't recall, I have this in https://github.com/openhardwarecoza/LaserWeb/blob/master/i/svg/js/SVGRenderer.js laserweb1. don't think it was actually used...

openhardwarecoza, 2016-05-02:
I'm seriously considering moving back to the SVGReader library we used in LaserWeb1 - snap.svg has some funnies :(

openhardwarecoza, 2016-05-09:
Right, svgreader.js in place - closing this issue for now.

I'll be adding some enhancements for SVG soon again (per color power/speed etc) but for now what I have in is a direct replacement for what we had with snap.svg

closed 2016-05-09 by openhardwarecoza