LaserWeb/deprecated-LaserWeb2 #30: X Y movement arrows suggestions change

cojarbi, 2016-04-21:
Is it possible to change the layout of this buttons to reflect X and Y in a cross pattern instead of side by side?

openhardwarecoza, 2016-04-26:
@cojarbi :  Like this?

[image]

cojarbi, 2016-04-26:
I think it's the best layout and makes it easier for everybody but will take a lot of screen realstate. Maybe something smaller like Chilli

cojarbi, 2016-04-26:
Ooooo I just saw you changed the whole thing.

openhardwarecoza, 2016-04-26:
[image]

Yip whole new feel (:

Just because i've been offline a lot, doesnt mean I've been idle (;

Go play on the live version (; and let me know what you think

Real estate is the biggest goal now + touch friendlyness: Those fabcreators come with a 1024x768 touch tft (;

cojarbi, 2016-04-26:
I'm sure even in your sleep you are making changes, upgrades, etc.

cojarbi, 2016-04-27:
Kind of complicated the Móvil experience https://imgur.com/vDcHWbF

openhardwarecoza, 2016-04-27:
True,  but the minimum resolution i'll cater for is a horizontal 7"
tablet.   Phones are just not the ideal platform

cojarbi, 2016-04-27:
Maybe phone just as a remote control. Much like the plugging for octoprint

cojarbi, 2016-04-27:
Can you make it so if press a button it opens the widget and if i press it again it closes it?

openhardwarecoza, 2016-04-27:
Phone as dongle:  i have some exciting (cheap) hardware on its way to
OpenBuilds to solve that one.  (;

Close widget: clicking View closes all widgets.   Coding it to close is a
possibility but i want the top toolbar to be more like a 'wizard'  guiding
users through opening,  setting up,  sending.  (not quite like i want it
just yet)

openhardwarecoza, 2016-04-27:
Gonna make the widgets draggable so you can see preview without them
obscuring your view

openhardwarecoza, 2016-05-02:
New UI in progress....

closed 2016-05-02 by openhardwarecoza