LaserWeb/deprecated-LaserWeb2 #31: Offset beween raster image preview and calculated raster

cojarbi, 2016-04-22:
Zoom in to the raster and youll notice that the calculated path has slight offset to the left. 
Is this:
affecting the actual location of the raster
its a bug from the image preview
Its part of the code originally intended by @nathanielstenzel 
Else?
[image]

openhardwarecoza, 2016-04-22:
Part  of that original 'raster to the point'  instead of 'from the point'
-  i just need shift the png/bmp/jpg over by spotSize

cojarbi, 2016-04-22:
Ok, understood. Look at this one. Its all over the place. Does the zoom/resolution slider affect the output if the image if not scaled properly to the dot size?
[image]

openhardwarecoza, 2016-04-22:
There are probably some rounding errors too.  I recall the bounding box
(gray size preview)  in LW1 did the same when you drag the mouse to set
spotsize but not if you used keyboard.  Noted that this does need some
enhancement though...  Busy week at work sadly :(

cojarbi, 2016-04-22:
Yeah Work. Why do we need that.

openhardwarecoza, 2016-04-22:
(;  thats a sadly true conundrum (;

nathanielstenzel, 2016-04-23:
Something that may help is if you can get it to render the path with a dot size. In other words, any line from point A to point B should end in circles around A and B and be as thick as the dot diameter. I had tried to get it to render like that before and failed.

nathanielstenzel, 2016-04-23:
That last image does look odd. Make sure you use parenthesis when doing math and structure it right when rendering. It kind of looks like someone got some sort of rectangle based math going on for the rendering and might have it structured wrong. That is just a guess though. If doing scaling of xmin and xmax, you should be doing scaled_xmin = (xmin- dotsize/2)×scale and scaled_xmax = (xmax+dotsize/2)×scale for the x bounds of the line, for example. Well, I hope that helps.

openhardwarecoza, 2016-04-23:
The preview is scaled as actual image dimensions × spotsize.    But the
gcode is actual size × spotsize + 1px left padding + 1/2 px right.   The
jquery ui slider also introduces error:  the preview is sized as a callback
from the slider.  But the gcode is sized by reading the slider val later.
Sometimes when you drag too fast the callback doesnt fire.  When you nidge
the slider with the keyboard it does.

What really needs to happen:   paperjs can resize a raster to any number of
dots we need.    Right now we let the user fuss with spotsize as zoom. But
actually spotsize is a constant on your machine -  we should adjust the
number of dots to change zoom.

openhardwarecoza, 2016-04-28:
Give it a try now @cojarbi

openhardwarecoza, 2016-04-28:
[image]

openhardwarecoza, 2016-04-28:
[image]
another test

cojarbi, 2016-04-28:
I'm getting the same results without scaling. When scaling it does not line up
[image]
[image]

openhardwarecoza, 2016-04-28:
Using mouse or keyboard on slider?  Mouse has inaccuracy

HakanBastedt, 2016-05-22:
Hi all,
I looked at this and know what happens.
The "slide:" event in $("#spotsizeslider").slider() (raster.js) seems not to be generated or caught for the final position. So the image on the screen and the slider value show one thing, when the raster button is pressed the last position of the slider is retrieved and used in g-code generation. Therefor the difference.

I duplicated the "slide: " function into a "change:" function (around line 85 in raster.js) and at least the images come out the same size. It is not the ideal solution though that anyone that tries it will find. But it shows were the problem is. 

Hopefully someone can take this info and find a real solution.

openhardwarecoza, 2016-05-22:
Yeah, i'l try escalate the bug upstream to jqueryui - but its an acceptable fix. Send pull when you get a chance?

openhardwarecoza, 2016-05-22:
nevermind, figured out what you meant:  Fire the "change" event from http://api.jqueryui.com/slider/

openhardwarecoza, 2016-05-22:
Thanks @HakanBastedt !

Size matches perfectly

[image]

cojarbi, 2016-05-22:
> Thanks @HakanBastedt !
>
> Size matches perfectly
>
> ![spot
> on](https://cloud.githubusercontent.com/assets/7695323/15453245/c9e84a92-200e-11e6-9e74-b4a29951c61f.PNG)
>
>
> ---

Awesome @hakannastedt

AY

closed 2016-05-22 by openhardwarecoza