LaserWeb/deprecated-LaserWeb2 #32: Raster image not working only getting one single line

Cinezaster, 2016-04-29:
I'm trying to do an engraving with the file / raster generator. I'll see in the other issues that the feature is working, but I actually never managed to generate gcode from a raster image. All I get is this:
[image]
One straight move.
What do I do wrong?

openhardwarecoza, 2016-04-29:
Weird one.  Post the file please?

openhardwarecoza, 2016-04-29:
Test file:

[image]

Result

[image]

Cinezaster, 2016-04-29:
This was the file.
[image]

I used your test file.
[image]
[image]

openhardwarecoza, 2016-04-29:
[image]

Ok so we eliminated file...

openhardwarecoza, 2016-04-29:
Aha, managed to replicate it:

If you forget to set a PWM max value in settings... it does this.  Need to add some defensive code to warn user to not to forget machine setup (;

[image]

Cinezaster, 2016-04-29:
Thx. I don't have my laser connected, that is why I didn't pay attention to the setting.

closed 2016-04-29 by Cinezaster