LaserWeb/deprecated-LaserWeb2 #33: Show file name

cojarbi, 2016-04-29:
Can we get LW to show the file name in use

openhardwarecoza, 2016-05-25:
(: Any tips on where?

cojarbi, 2016-05-25:
As you open files on the top left, thats where if feel it's natural, so it could also benefit the Google Drive load you can place the name just bellow the "open from drive"

openhardwarecoza, 2016-05-25:
[image]

Figured out that I can add more things onto the 3D viewer... Might be crazy, but I kind of like the idea of using the canvas for this sort of "info" like filenames, job time estimates, etc

cojarbi, 2016-05-25:
I did no suggest there because though was not possible. 

Look very good.  Migth also suggest on the top left of the canvas as a "tab"  as all GUI this days use that in one way or the other

AY

openhardwarecoza, 2016-05-25:
Ahh!   You just cracked ny biggest gui worry!  How do display multiple
layers and set each's cam job!!!!  By adding tabs!  Why didnt i think of
that!   You just gave me the gift i've been looking for week!  As you add
jobs,  i add tabs to the top.   Allows user to switch between files and
tweak each 'layer',   when we run final job,  we iterate through each
tab.   Each tab can have its own cam,  own gcode textarea,  etc.   One
final tab shows everything stacked.  F%ck this is going to look awesome.

cojarbi, 2016-05-25:
Exelent. Full speed ahead. I did no suggest anything there bacause of the many times you told me that area was untouchable. 

Defenetly tabs is the best way to go.  Clean,  simple and today everybody are use to them. 

AY

openhardwarecoza, 2016-05-25:
Thanks again.  Tabs take so little space,  and from a flow perspective,
its just perfect.  Makes sense.  I like UIs that make sense

cojarbi, 2016-05-26:
Regarding the name dont know if your seen but opening gdrive files does not show the file name

openhardwarecoza, 2016-05-26:
I know,  totally different set if functions to the normal button

openhardwarecoza, 2016-05-29:
Closed,  since filenames will be the tabs too

closed 2016-05-29 by openhardwarecoza