LaserWeb/deprecated-LaserWeb2 #35: Load settings from controller into laserweb

Cinezaster, 2016-05-02:
In grbl you can receive the size of your cnc by asking the settings. The grbl object already contains the size of the cnc after you send `$$` command, by storing all the data into it's memory.

It could or should set the size of the cnc in laserweb. 

Since I'm gonna start implementing the TinyG & TinyG2 object, we should think about a flow to retrieve data from our controllers to have a validation layer in laser web that generates better gcode that does not try to go faster or is not going to hit the limits of the machine.

openhardwarecoza, 2016-05-02:
Agreed.  Good idea.

openhardwarecoza, 2016-05-02:
Conversely we can push the bed size to the machine when changing it in the
gui too (;

Cinezaster, 2016-05-02:
OfCourse, that would be the whole point. But then we need our controller communication layers.

closed 2016-09-27 by openhardwarecoza