LaserWeb/deprecated-LaserWeb2 #36: TinyG - TinyG2 tested communication layer

Cinezaster, 2016-05-02:
Creating the communication layer between laserweb2 and TinyG - TinyG2
* [ ] handles all communication and parses all the messages
* [ ] stores all controller settings into TinyG object
* [ ] make it tested
* [ ] pre installed command for homing, resetting, ...

openhardwarecoza, 2016-05-02:
Don't know if this would help but @giseburt wrote https://www.npmjs.com/package/tinyg which could contain some value

Cinezaster, 2016-05-02:
It is mainly for the serial connection and it is a low level implementation it doesn't abstract the communication from the tinyG into interpretable data.  
Nicely written library, he put in a lot of work into it and not only in the code but also the documentation.

Cinezaster, 2016-05-03:
On connecting with the device we need a command we can send to recognize which device is connected. TinyG2 is silent on connect. 

This command must be a unique command for tinyG and/or TinyG2 (their behaviour seems the same).
I suggest: `{sys:n}` this returns the [system group of TinyG](https://github.com/synthetos/g2/wiki/Configuration-for-Firmware-Version-0.98#system-group)

openhardwarecoza, 2016-05-03:
Since the user has to select the Buffer for Serial Port Json Server
already,  we could use that as an 'assumption' that it is a tinyg?

In laserweb1 I did this
https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L174-L177

Chilipeppr also checks for $fb (firmware build)

Cinezaster, 2016-05-03:
The buffer select could indeed be the solution.

Never the less, I like to import all data from the device into laserweb. For later use
The `$fb` command returns a less parseable response compared to the `{fb:n}` command which returns json.

I found a Arduino Due and flashed it with TinyG2. Would be great if someone could send me a list of all the settings of a realistic machine. Don't want to spend time configuring one.

openhardwarecoza, 2016-05-03:
I'll nudge the guys on Google Plus again...

PS: no objection against $fb vs $fb:n

closed 2016-09-27 by openhardwarecoza