LaserWeb/deprecated-LaserWeb2 #37: Laserweb api

Cinezaster, 2016-05-02:
We might have to think about a laserweb api. 
Otherwise this will become hard to maintain.
Here some quick ideas.
```
laserWeb.settings = {...} // object containing all the settings
laserWeb.settings.set('x', 15000) // a way to set the settings, this way we can also trigger events
laserWeb.settings.set('homingCommand', '$H') 
laserWeb.controller.home() // home your machine

laserWeb.raster
laserWeb.dxf
laserWeb.gcode
laserWeb.controller

laserWeb.trigger('WPos',{x:231,y:541,z:9})
```
This is more testable :-)
The UI would hook into the laserWeb object

openhardwarecoza, 2016-05-02:
Only reason i havent done that is that my self taught skills havent gotten
there yet :(

Go for it (and hopefully with a few comments on commits i'll skill up along
the way)

Cinezaster, 2016-05-02:
We might have to branch it off, while working on it. The gh_pages branch strategie might not be that ideal.
Again, this will not be for today :-p and we'll have to start the discussion first, about the api, then write the documentation and tests for it and then start to write some code.

mkeyno, 2016-05-05:
hi @openhardwarecoza peter I've always follow up your great work on laserweb2 if you may know   @luc-github [at last update new ESP link for 3D printer](https://github.com/luc-github/ESP8266)   and It work very well and fully transparent for 3D printer serial  & webserver , however I've tried to add your files & lib to his, so we could have both CAD & wireless link features, however I've find ESP so powerless for such huge amount of laserweb2 files ,  so I've been thinking maybe it is best that you convert your javascripts codes to API or Plugin for browser , so when I install it just call it by ESP webserver and can process and share all need for any CAD feature 
BR

openhardwarecoza, 2016-09-27:
Finally happening in LW4 (:

closed 2016-09-27 by openhardwarecoza