LaserWeb/deprecated-LaserWeb2 #42: Google drive files

cojarbi, 2016-05-05:
Selecting SVG produces the following " ReferenceError: drawSVG is not defined (http://openhardwarecoza.github.io/LaserWeb2/js/googlesignin.js on line 123)"

DXF load without issues

BMP and jpg don't show at all

Maybe show a pop-up like touchpad to select file im afraid that a long list will move the whole left side of the interface down
[image]

openhardwarecoza, 2016-05-05:
It still has to move to a tab of its own

yuusuf-sallahuddin, 2016-05-24:
[image]
Issue here for me with the G Drive integration whereby it seems to list only some of the files in my G Drive. I've got a tonne of folders in my G Drive & I'm not seeing any way to navigate the folder structure.

openhardwarecoza, 2016-05-24:
I'll add folders later.  For now i am filtering only files we support
(dxf,  svg,  bitmaps)

openhardwarecoza, 2016-05-24:
Folders integrated. Just updating here for someone looking later perhaps...

closed 2016-05-24 by openhardwarecoza