LaserWeb/deprecated-LaserWeb2 #43: Starting LaserWeb2

Mad-Onion, 2016-05-05:
Maybe I'm just too stupid, but how do I start LaserWeb2?
The instructions from LW1 don't work:
´´´
C:\Users\...\LaserWeb2>node server
module.js:338
    throw err;
          ^
Error: Cannot find module 'C:\Users\Tobias\Entwicklung\Microcontroller\LaserWeb2\server'
    at Function.Module._resolveFilename (module.js:336:15)
´´´

Best regards
Tobias

openhardwarecoza, 2016-05-05:
Go to http://openhardwarecoza.github.io/LaserWeb2 as per LaserWeb2's
readme.   #rtfm (;

openhardwarecoza, 2016-05-05:
Just read the readme...  Note LaserWeb2 is in early-mid development.  Svgs has issues,  anything other than grbl and smoothie hasnt had much testing etc. If you are a developer or clued up user,   help out with LaserWeb2.   If you are a normal user go to laserweb1.

closed 2016-05-05 by openhardwarecoza

Mad-Onion, 2016-05-05:
I found the link to the demo, but I want to start it locally in my windows-box, because I have the "bad number format"-grbl-problem.

So I cloned the LW2-repository and tried to start it the way it was done in LW1 - which didn't work :(
I didn't find anything in the readme regarding how to actually start the program...

Gustavvr, 2016-05-05:
It is not a standalone server you need to run it in a web server. Apache or similar. I don't think @openhardwarecoza is ready to release docs yet. PM me and I'll try to give you a hand offlist

openhardwarecoza, 2016-05-05:
No need for apache.  Literally: go to
http://openhardwarecoza.github.io/LaserWeb2    thats it.  There it is.  Use
it!  I spent a lot of effort getting it working in github pages...

Just use SPJS from the chilipeppr project as a serial bridge...

cojarbi, 2016-05-05:
Running LaserWeb2 locally:

(nb: make sure you keep up to date,  i'll ignore support requests if you run local but dont tell me you are 'git pull'  up to date) 

1. Install nodejs
2.  npm install -g http-server
3. Git clone LaserWeb2 to say c:/LaserWeb2
4.  Run http-server c:/LaserWeb2 

﻿

Gustavvr, 2016-05-05:
@openhardwarecoza Wow this is an amazing plan. I get what you're doing now.  Yeah, Totally not worth running my own server.

This is really great. Thanks,

openhardwarecoza, 2016-05-05:
At least at this phase (:  i mean theres 3-4 updates a day.   If a user
(^)  cant even read a readme,  how heavy is the support load going to be
since they wont keep up to date (;

Mad-Onion, 2016-05-05:
I believe I was thinking to complicated. Never thougth that a browser-app would communicate to a serial-port via a program running on my local host... fancy shit! ;-)

openhardwarecoza, 2016-05-05:
Dobt forget to play with the Settings tab first