LaserWeb/deprecated-LaserWeb2 #46: Home control does not work with Marlin

funinthefalls, 2016-05-13:
All the jog buttons work, however I cannot make it home using the home button. Also the abort job button only works about 50% of the time for me.

openhardwarecoza, 2016-05-13:
Home:.. Did you remember to configure the settings,. Gcode,  Homing
Sequence?

Abort: clears the spjs queue,  but there's still a buffer on the controller
that meeds to finish first

funinthefalls, 2016-05-13:
Will save the settings again and give it a try. Hopefully that is all that was wrong. On a different subject, I am getting very strange results from the raster function. I will post the gcode it creates along with the original bitmap. Other that that I am really liking the workflow in LaserWeb 2. It is turning into a great piece of software Peter, you deserve a great deal of praise for your efforts.

openhardwarecoza, 2016-05-13:
Yip send over files (;

openhardwarecoza, 2016-05-13:
If you see weird burn patterns - that's why i hate marlin.  Doesnt have
power as function of acceleration: so acceleration and deceleration has
darker burns :(

funinthefalls, 2016-05-13:
Here are the files.

[image]
[bday.gcode.txt](https://github.com/openhardwarecoza/LaserWeb2/files/262583/bday.gcode.txt)

funinthefalls, 2016-05-13:
TYhe home issue was due to not saving the home codes properly....dumb mistake on my part.

openhardwarecoza, 2016-05-13:
Transparent BG PNGs confuse the heck out of paper.js when I ask it to report the "gray level" of the pixel.   Merge down and export as flat preferable with a white background

funinthefalls, 2016-05-13:
OK, will give it a try.

funinthefalls, 2016-05-15:
That fixed it,thanks! Closing the issue

closed 2016-05-15 by funinthefalls