LaserWeb/deprecated-LaserWeb2 #48: Keeps previews gcode

cojarbi, 2016-05-15:
So i open a file, generated the code the open another file and the previous gcode is still there. 

More over when you go to convert to generate a new code it keeps showing and accumulating the previous one. Is this by design? It does execute the correc one.
[image]
[image]

openhardwarecoza, 2016-05-24:
Is this still an issue?

cojarbi, 2016-05-24:
It does. Just tested svg. No dxf at hand

openhardwarecoza, 2016-05-24:
Okay so #59 relates... Need to fixup threeCleanUp function...

openhardwarecoza, 2016-05-25:
Okay testing OK on my side, waiting for your confirmation

cojarbi, 2016-05-25:
So it does clear the previous image but still leaves the previous in the conver
[image]
t tab

closed 2016-09-27 by openhardwarecoza