LaserWeb/deprecated-LaserWeb2 #52: Smoothie chokes with Marlin buffer

cojarbi, 2016-05-22:
This is a complete dump of the terminal running a complete job. 

Tested 57600, 115200, 230400 and 250000 ( this last one did not connect )

Always chokes on 300 aprox lines remaining in the queue
[Smoothie with marlin buffer.zip](https://github.com/openhardwarecoza/LaserWeb2/files/276623/Smoothie.with.marlin.buffer.zip)
.

openhardwarecoza, 2016-05-25:
New comms layer progress:  https://github.com/Cinezaster/serialPortJsonClient

openhardwarecoza, 2016-05-30:
Reopening, looking at the log, I dont think the rearchitected js client side will address, this looks more like we need to more work on the marlin buffer (i must also remember to rename it reprap buffer) or we may need to add a Smoothie buffer.  

I can see the SPJS queue going down (from line 6882 in your log)

2016/05/22 17:07:06 bufferflow_marlin.go:159: We should NEVER get here cuz we should have a command in the queue to dequeue when we get the r:{} response. If you see this debug stmt this is BAD!!!! worries me a little - may mean we have something wrong in the marlin buffer code on SPJS (@johnlauer)

After a few moves it looks like thge buffer size goes over the 128 we specified in the marlin buffer, and thus SPJS pauses?

I assume we arent dequeuing the command from the 128 char queue?  @johnlauer?

openhardwarecoza, 2016-05-30:
Looking at the old marlin buffer go code, we also have an init line waiting expressly for the string "marlin" - probably also an issue if we try to use the marlin buffer with other reprap firmwares like repetier and smoothie

openhardwarecoza, 2016-05-30:
@raykholo

johnlauer, 2016-05-30:
That means your syncing is off. For each affirmative response from your controller you should have the command in the spjs queue so it can link the two up. If you get that error something is wrong with the syncing.

raykholo, 2016-05-30:
I'm in the CP grbl workspace with one of my Smoothies.  Initial connect good. CP doesn't recognize initial 'Smoothie' feedback on board connect, but that doesn't kill anything. Can jog around fine.  

[image]

I hit run on the default job, queue goes up to 581 and stays there.  Board may have even frozen.  

[image]

Is this roughly the same as what is being experienced in LW?  If so, SPJS buffer issue?

cojarbi, 2016-05-30:
@raykholo that's what happens to me.  It fills the buffer but then once the board has reached that point it seems it did not try to resend new lines and the operations stops until further info has arrived. At the end it freezes the board and im forced to reset it

raykholo, 2016-05-30:
Precisely. Sounds like a SPJS buffer issue to me. Let's debug further. 

Ray

HakanBastedt, 2016-05-31:
Fwiw I run a Marlin board with the marlin buffer and the flow control appears to fail.

openhardwarecoza, 2016-05-31:
I dont have a smoothie connected to a machine at the moment, but heres a run on a smoothie, with no motors connected: (100x200px raster)

SPJS 1.92 on Windows 10:

FWIW looking at the incoming position feedback I didnt see it stalling.  It ran all the way through, buffer flushed down to 0...

[log.txt](https://github.com/openhardwarecoza/LaserWeb2/files/290805/log.txt)

openhardwarecoza, 2016-05-31:
@johnlauer fyi, when you click play I do

```
  var g;
    g = document.getElementById('gcodepreview').value;
    sendGcode(g);
```

and 

```
function sendGcode(gcode) {
    console.log('Being asked to send ', gcode)
    if (isConnected) {
        var gcode = gcode;
        wsSend('send ' + $('#port').val() + ' ' + gcode);
        console.log('send ' + $('#port').val() + ' ' + gcode)
    } else {
      var gcode = gcode;
      printLog('Could not send ' + gcode + ' - NOT CONNECTED', errorcolor)
    };
};

```

Where wsSend is:

```
wsSend = function(msg) {
    console.log('Being asked to send ', msg)
    if (isWsConnected) {
        conn.send(msg);
        console.log('Sending: ', msg);
    } else {
        console.log("Tried to send message, but we are not connected to serial port ajax server.");
    }
};
```  

which came from the CP widgets spjs code... 

Unless thats the wrong way to send a bunch of lines?

 g = document.getElementById('gcodepreview').value;  =   a textarea containing the lines of gcode split by \n

openhardwarecoza, 2016-05-31:
I think the error about 'We should not get here' is in the queryrpt not in the queued gcode.  @johnlauer i may need your help here.
.... arrLines:[ok C: X:19.1000 Y:0.1000 Z:0.0000 E:0.000  ]
2016/05/31 11:56:08 bufferflow_marlin.go:120: We have data lines to analyze. numLines:2
2016/05/31 11:56:08 bufferflow_marlin.go:133: Working on element:ok C: X:19.1000 Y:0.1000 Z:0.0000 E:0.000 , index:0
2016/05/31 11:56:08 bufferflow_marlin.go:159: We should NEVER get here cuz we should have a command in the queue to dequeue when we get the r:{} response. If you see this debug stmt this is BAD!!!!
2016/05/31 11:56:08 bufferflow_marlin.go:164: Marlin just completed a line of gcode
2016/05/31 11:56:08 bufferflow_marlin.go:209: OnIncomingData() end.

In response to M114 - smoothie returns "ok C: X:19.1000 Y:0.1000 Z:0.0000 E:0.000"

And in the code we borrowed from the Grbl buffer we expect the returned response to be "ok" - ie would having chars next to ok, cause that "bufferflow_marlin.go:159: We should NEVER get here cuz we should have a command in the queue to dequeue when we get the r:{} response. If you see this debug stmt this is BAD!!!!"

johnlauer, 2016-05-31:
That code looks fine at first glance, but that doesn’t mean much. Code like serial sending can only be debugged while you’re seeing data flow across it.

openhardwarecoza, 2016-05-31:
(; can we donate you a Smoothieboard?

johnlauer, 2016-05-31:
What do you do to detect a line is done and is that consistently getting detected?

johnlauer, 2016-05-31:
Thx for the offer, but I don’t have time to do it and if I do have time I have a much larger priority list on CP items.

HakanBastedt, 2016-05-31:
It looks very much like the number of "ok"s are not counted correctly. I suspect a lot 
that all responses from the board count as "ok" even for the regularly asked for coordinates.

I took out my logical analyzer and recorded the serial traffic over usb.
I have formatted the data into an understandable format, I attach the librecalc ods file
[ch67s.zip](https://github.com/openhardwarecoza/LaserWeb2/files/291923/ch67s.zip)
You can check that the buffered lines stays nicely below 10 but only if the not-"ok" responses are counted as "ok". Otherwise the buffered lines are like 58 and if I see how the laser reacts with strange moves and all such, then I think of memory overrun.

I using spjs 1.92

openhardwarecoza, 2016-05-31:
The what gets checked as completed,  is still exactly as @johnlauer helped
me in the session https://youtu.be/qZ-GtkXUc40 -  i havent changed anything
since that original session.

@HakanBastedt awesome that confirms what i suspected above.

So that means,  we need to extend the regex on
https://github.com/chilipeppr/serial-port-json-server/blob/master/bufferflow_marlin.go#L50
to accomodate the single line responses?  And extend init to include
smoothie,  repetier,  and non-turnkeytyranny marlin init strings.  Probably
also our own error responses for the various firmwares

#Firmwaredevswhyunotstandardise!

HakanBastedt, 2016-05-31:
Or is it the rptqueryloop? Doesn't seems to wait for clearance on serial data? I may be wrong, this "go" is a new thing.

openhardwarecoza, 2016-05-31:
The go thing is new to us all.  Theres only one expert here (;  on that
one.

HakanBastedt, 2016-05-31:
Haha, well I read the comments and it says that rptQueryLoop() sends data outside the normal flow. I know grbl has that but Marlin doesn't which is a pity sometimes. So all commands go into the one and only buffer for further processing - except for m112. If that is this indeed the case in rptQueryLoop() than this need to be changed to send in M114 as a "normal" command. At least that is what I would have tried if I could.

openhardwarecoza, 2016-05-31:
Ok so i setup a golang build environment again, i'm going to do a few things:

1.  Rename "marlin" to "reprap" - all the reprap firmwares reasonably talk the same gcode...

2.  I am not a regex jocky, but i'll play with a few online regex testers and known strings to make sure we write regex that can match m114 succesfully for marlin OR smoothie OR repetier...

Currently

```
	b.reNewLine, _ = regexp.Compile("\\r{0,1}\\n{1,2}") //\\r{0,1}
	b.ok, _ = regexp.Compile("^ok")
	b.err, _ = regexp.Compile("^error")
	b.initline, _ = regexp.Compile("")
	b.qry, _ = regexp.Compile("\\?")
	b.rpt, _ = regexp.Compile("^X::")

```

3.  I can already see that ^ok  should be ^ok$  (else it will match for ok C: X:19.1000 Y:0.1000 Z:0.0000 E:0.000	 as well.  
But the b.rpt shouldnt be ^X: but ^ok C: for smoothie (i dont have the marlin and repetier responses for m114 on hand...

Its going to be painful but we'll literally need make sure that ok, err, rpt, etc are all caught correctly :(

openhardwarecoza, 2016-05-31:
... and cross platformy...

openhardwarecoza, 2016-05-31:
@Cinezaster  ^ spanner in your plans above

openhardwarecoza, 2016-05-31:
@HakanBastedt can you try and gather all the expected responses from your Marlin fork?  (What does it say on startup, what denotes an error, what denotes a ok:, how does M114 look, etc

Similar to what I ha in LW1 https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L371-L389

openhardwarecoza, 2016-05-31:
and javascript parsing is much easier than regex :(

openhardwarecoza, 2016-05-31:
**Smoothie:**

Init = 
```
Smoothie
ok
```

M114 = 
`ok C: X:0.0000 Y:0.0000 Z:0.0000`

G0 X100 =
`ok`

S0.6 = 
`ok`

openhardwarecoza, 2016-05-31:
Now I think we should be

```
	//define regex
	b.reNewLine, _ = regexp.Compile("\\r{0,1}\\n{1,2}") //\\r{0,1}
	b.ok, _ = regexp.Compile("^ok$")
	b.err, _ = regexp.Compile("^error")
	b.initline, _ = regexp.Compile("^(Smoothie|echo:Marlin)$")
	b.qry, _ = regexp.Compile("\\?")
	b.rpt, _ = regexp.Compile("^(ok C:|ok X:)")
```

openhardwarecoza, 2016-05-31:
Alright, compiled and ran, not getting a "We shouldt get here" anymore

HakanBastedt, 2016-05-31:
Init
`Printer reset detected - initalizing`
`start`
`echo: External Reset`
`Marlin 1.1.0-RCBugFix`
`echo: Last Updated: 2016-04-27 12:00 | Author: (none, default config)`
`Compiled: May 26 2016`
`echo: Free Memory: 2526  PlannerBufferBytes: 2640`
`echo:V23 stored settings retrieved (398 bytes)`

M114 =

` X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0 Y:0 Z:0`
`ok`

G0 X100.0
`ok`

S0.6
`echo:Unknown command: "S0.6"`
`ok`

There are also a few
`busy: processing`
`busy: paused for user`
`busy: paused for input`
They show up in the X, Y, Z DRO fields which is fun at first.
The "busy: processing" can show up anytime - it has to do with commands that take a long time but can't really be predicted.

openhardwarecoza, 2016-05-31:
Attached windows binary - please test

(rename ,zip.txt to .zip  - github is rejecting .zip and .exe)

[serial-port-json-server.zip.txt](https://github.com/openhardwarecoza/LaserWeb2/files/292158/serial-port-json-server.zip.txt)

HakanBastedt, 2016-05-31:
"ok"

openhardwarecoza, 2016-05-31:
Right lets move this to the SPJS repo (;

openhardwarecoza, 2016-05-31:
@HakanBastedt ... "ok" instead of "^ok$" ?

HakanBastedt, 2016-05-31:
Yes. or I mean no. The motion stopped early now with a similar error. Didn't use the logic analyzer but problem is there.
Can you perhaps switch off the M114 all together to rule out that?

openhardwarecoza, 2016-05-31:
Check the readme on spjs,  and setup a local environment so you can build
binaries quickly.  Uploading from here on my slow connection is taking
forever.  I am busy pushing the git commit to
openhardwarecoza/serial-port-json-server  (i made you contributors too)

HakanBastedt, 2016-05-31:
I feel really unqualified for go but yes I can give it a try.

raykholo, 2016-05-31:
+1 for liteide, very helpful for running a go app without having to type
stuff in terminal yourself.

openhardwarecoza, 2016-05-31:
Me too.  But i am just replacing the regex strings.  Its relatively easy
luckily.  The readme works for setting up build environment.  And go build
does the build

raykholo, 2016-05-31:
What Peter and I are thinking:  "I should just stay behind Eagle where I belong.  But I can google, copy, and paste things until it 'works' so let me keep trying that. .."

openhardwarecoza, 2016-05-31:
Hey 99.2% of laserweb is copied from either someone elses code, or stack over flow.  I can't code to save my life!

openhardwarecoza, 2016-05-31:
Considering the advanced maths @HakanBastedt contributed to the Raster code, Go should be peanuts (;

raykholo, 2016-05-31:
As long as no one has an allergic reaction to Go...

cojarbi, 2016-05-31:
That's was really funny

AY

HakanBastedt, 2016-05-31:
Wow, go is quite neat! 
Yeah, it is the rptQueryLoop() that is the problem. It injects M114 commands but the "ok" from that command is not taken care of. My laser has run for 10 minutes and it just works perfectly fine now. Nice, nice.

Can someone with great skills in go modify the rptQueryLoop()? The M114 commands has to go into the normal serial stream and the "ok" must be taken care of, just like any other command in the serial stream.

openhardwarecoza, 2016-05-31:
Does marlin return ok\nX:.... ? (line in between?)

HakanBastedt, 2016-05-31:
See above. No newline, just X: ... \nok\n

openhardwarecoza, 2016-05-31:
Ahh thanks!  I forgot that you already pasted the outputs,  sorry.

We can handle that in the b.rpt query.   Just needs to match from ^x:....
To ok$ so the b.rpt eats the OK.  Time to play regex

openhardwarecoza, 2016-05-31:
Something like
http://stackoverflow.com/questions/20173472/does-go-regexps-any-charcter-match-newline
to match over the newline to the ok$

HakanBastedt, 2016-05-31:
Job completed, no errors. Great.
Referring to the original problem statement, I assume that this fix will allow Smoothie to work on marin buffer.
I am lost on go, but if the "ok" can be eaten with the "X:" line, then it should work.

HakanBastedt, 2016-05-31:
But I am not sure, I think it is a dangerous practice for rptQueryLoop() to send data and just expects the board to handle it. Ah well, let's try you solution with regexp.

wolfmanjm, 2016-05-31:
FYI you cannot use GRBL or Marlin buffer control with smoothie, it won't work as Smoothie is using a CDC USB serial driver protocol and not a uart. If you try to guess the buffer size it will fail.
The rule with smoothie is send a line, wait for ok, send next line. That is the protocol. (bCNC does this now and it works). This does not cause pauses as it would in GRBL or Marlin because the serial stuff is totally different. There is no way to know how big the buffer is, so you need to follow the send, ok, send...rule

johnlauer, 2016-05-31:
Well, then your best bet is to do what I did with the TinyG G2 buffer which is also a USB protocol. I just assumed a 20,000 character buffer which is more than enough safety margin since USB has a huge buffer. What you don’t want to do though is fill it up too high such that prioritized commands like feed hold get way backlogged in the queue. So I would say just go with a 1,000 character queue to be safe.

wolfmanjm, 2016-05-31:
@johnlauer  No that will not work with smoothie. As I said the ONLY protocol that will work is send... wait for ok... send, if you send without waiting for an ok you will get problems I guarantee it. There is NO serial buffer so sending without waiting for ok will simply not work.

johnlauer, 2016-05-31:
Hmm. Ok.

wolfmanjm, 2016-05-31:
to be more specific what WILL happen is the USB flow control will kick in and your send will block, which will cause deadlock as you will probably not be able to clear it by reading the backlogged oks, (unless you have a separate read and send thread) In addition if the USB flow control did not kick in it would cause other potentially random issues, as the ok response is hooked into the planner on smoothie and if the planner is busy it won't send an ok until it can plan the next line. So please follow the protocol I outlined above, that works well, and is used by pronterface, octoprint, bCNC and most other hosts.

raykholo, 2016-05-31:
Wow, I think this explains my issues with Repetier Host, if they didn't implement that properly.  I can send single commands like jog and heat up fine, but get freezes within the first 30 seconds when I run a job.

wolfmanjm, 2016-05-31:
@raykholo AFAIK there is a setting in rep host to make it do the correct thing.

openhardwarecoza, 2016-06-01:
Well that explains why the dumb buffer in LaserWeb1 worked so well for snoothie.  All it does is wait for ok,  then send.    But that approach fell flat fo Marlin and Grbl.  Never got get to the bottom of it,  but to grbl data got mangled.  For marlin we seemed to get an OK when the command was recieved (and thus could flood the buffer if we added moves before the move was executed.  

The long and the short is we need to sort this out.  And the only way to do this is to have all the firmware role players talking to each other.   Being the host developers we are stuck in the middle of everyone's different way of doing things.  I could say 'screw grbl,  screw marlin,  i only support smoothie'  Or i could say 'screw serial,  we only support wifi sd cards and smoothie ethernet  -  but that will make my project disappear in no time at all.  

I really hope we can all come together and discuss a' how best to talk to the firmware.   I also hope that we can integrate it into SPJS with @johnlauer's help.  I know he hates anything thats not a tinyg (; but the world wont change anytime soon.  There are always going to be different camps of fans.  And the best we can do is to support everything,  but in a way that plays the firmware dev's hands in a nice way.  Even if it means we need to put the code into firmwares with pull requests to add a little more standardised responses.  I would gladly do that.   I know a lot of the host developers @Octoprint @repetier @kliment (tag the head marlin guys please?)  would also be glad (since just as we have spjs fans,  we also have repetier host fans,  Octoprint fans,  Pronterface fans.  Of we all work together,  who wins.? Everyone! Host and firmware devs

HakanBastedt, 2016-06-01:
I had a look and the buffer_marlin isn't suitable for Marlin either, most likely it may work with Repetier.
Reason is that it (everything is as far as I can understand) tries to fill the boards 127 byte buffer. That is perfect for Repetier because it has a 128 bytes ring buffer. But Marlin doesn't work that way, it has 4 lines buffer for serial data. Like 100 bytes each, but still only 4.
One way to combine this is to have a method that sends x lines and keep in sync with "ok"s, for Smoothie x=1, for Marlin x=4. And that M114 is injected into the normal serial stream and treated as a normal command.

[53 more replies]

closed 2016-06-02 by openhardwarecoza