LaserWeb/deprecated-LaserWeb2 #63: Recalculating rasters

funinthefalls, 2016-05-25:
If I open a bitmap and generate the raster, then change the dpi and re generate the raster, it overlays the new view over the original view.

openhardwarecoza, 2016-05-25:
Duplicate of #48

funinthefalls, 2016-05-25:
It is not the gcode that was the problem, it was the preview image. Not sure if that makes a difference to you.

openhardwarecoza, 2016-05-25:
I tested twice, can't replicate...

Check if its still happening? Screenshot might be helpful if it does

cojarbi, 2016-05-25:
Not sure if this applies but, raster and then load an svg
[image]

openhardwarecoza, 2016-05-25:
The svg cleanup is your old ticket i am quite sure

cojarbi, 2016-05-25:
Ok so i believe @funinthefalls is talking about this, but it shows the preview over/behind the the old raster preview but once generated it disappear  
[image]

openhardwarecoza, 2016-05-25:
Oh right,  yes thats whats going on.   I show the gcode.  Until you
generate gcode...  So you want me to clear the gcode as soon as you change
the DPI?

funinthefalls, 2016-05-25:
@cojarbi That is exactly what I was seeing. I think clearing the gcode when changing the dpi would be nice, if it does not hinder performance. Seeing the double images might confuse us lesser mortals. :)

cojarbi, 2016-05-25:
Think of it for a second, having the previous code gives you a compare over the previous size,  etc. Could be beneficial maybe on small complicated arts. It does fell weird I agree.

cojarbi, 2016-05-25:
Forget it, would only help when enlarging over the previous one not the other way arround

cojarbi, 2016-05-25:
Unless there is some black magic the will bring the new one forward or back on the screen.  This could be good when the live feed from the camera is working for testing perfect placement

funinthefalls, 2016-05-25:
Will test later today, will be out of my office for most of the day.

openhardwarecoza, 2016-05-25:
Ok clearing gcode when you change dpi...

closed 2016-05-25 by openhardwarecoza

funinthefalls, 2016-05-25:
Thanks Peter