LaserWeb/deprecated-LaserWeb2 #65: Problems with raster creation

funinthefalls, 2016-05-29:
Since the last merge, files that generated the raster and preview and gcode in less than a minute are now taking much longer (5 -10 minutes to complete the process, or even causes a hang sometimes). The raster seems to generate just as quick, but creating the gcode and displaying the preview seems to be where the bottleneck is. Not sure what has happened, but it definitely feels broken.

cojarbi, 2016-05-29:
I get the same

funinthefalls, 2016-05-30:
Funny thing is it works at http://hakanbastedt.github.io/ without any issues.

openhardwarecoza, 2016-05-30:
Sorry, the word 'fix' in the commit name autoclosed the issue.

I manually pasted over the two files from https://github.com/HakanBastedt/HakanBastedt.github.io/  
Check if it resolved your issues?  (Seeing as you say it works without issue on  http://hakanbastedt.github.io/ it should) 

@HakanBastedt ^ (;

openhardwarecoza, 2016-05-30:
Though looking at the generated diff off the two pastes, I dont see anything off.  (I only added some unfinished G7 code lol)

funinthefalls, 2016-05-30:
Whatever you did, it fixed the problem.

HakanBastedt, 2016-05-30:
I take it that it works now?

funinthefalls, 2016-05-30:
Yup

funinthefalls, 2016-05-30:
Having problems once generating the gcode again, system hangs for up to 3-5 minutes, process a little more, hangs for a few minutes, process a little more and so on. Was working after Peter made a fix earlier, looks like the Y resolution increase has done something to it.

HakanBastedt, 2016-05-30:
Strange, just tested here on openhardwarecoza.github.io and it worked nicely for me. Is it a particularly large image or so? Is it better when you start with a new freshly started browser?
I do 2000x2000 without problem, and I am sure I can do much larger than that. 

The Y resolution should not influence this, the same amount of G-code is generated now.

funinthefalls, 2016-05-31:
It is now working for me. Not sure what the issue was, but this can be closed now.

commit 2c508393 (2016-05-30): Attempt to fix #65
commit ed89b4f5 (2016-05-30): Attempt to fix #65
closed 2016-05-31 by funinthefalls