LaserWeb/deprecated-LaserWeb2 #67: Add Jog feedrate setting

SloMusti, 2016-06-03:
Been missing this one for use with CNC machines and non-laser applications mostly, would be very useful for testing things as well as moving a machine about.

Should this already be implemented, please let me know.

openhardwarecoza, 2016-06-03:
I already have Settings - Gcode -> Travel Moves as a configurable value but havent done all it should be...

If you have a few minutes, can you help me out by 

add a Travel Moves (z) below https://github.com/openhardwarecoza/LaserWeb2/blob/gh-pages/index.html#L594-595  - most machines need a slower feed on Z (or they forgot to configure max feedrate in firmware for z).  Change the ID to rapidspeed and rapidspeedZ for example for the second field

 <label for="endgcode" class="control-label">Travel Moves (mm/min): XY</label>
                                <textarea class="form-control numpad" id="rapidspeed" placeholder="2000"></textarea>

 <label for="endgcode" class="control-label">Travel Moves (mm/min): Z</label>
                                <textarea class="form-control numpad" id="rapidspeed" placeholder="2000"></textarea>

Add the ID of the new fields (rapidspeed and rapidspeedZ) to https://github.com/openhardwarecoza/LaserWeb2/blob/gh-pages/js/localstorage.js#L7 so it gets saved and loaded

Update the commands in https://github.com/openhardwarecoza/LaserWeb2/blob/gh-pages/js/jog.js to for example

```
var feedRate = $('#rapidspeed').val();
sendGcode('G91\nG0 X'+ dist + 'F' + feedRate + '\nG90\n');
```

openhardwarecoza, 2016-06-03:
PS I know some may complain about hiding it in the settings tab - but on the 1024x768 UI on the onboard LCD of the #FabCreator) we cant spare more space on the Jog tab on the front ui... Though i find i rarely change feedrates.  I actually set my homing, default and max feedrates in firmware...

SloMusti, 2016-06-03:
The argument why jog rates need to be in UI for CNC machines is, because you often do some manual machining this way, setup or testing. Even with laser one might want to manually jog and cut with it, usually with jog keyboard on the controller.

Will have a look at this shortly, tnx for fast and awesome reply.

openhardwarecoza, 2016-06-03:
I hear you...  Show me where it fits on 1024x768 while still bein touch
friendly and i'll merge it

SloMusti, 2016-06-03:
Agreed, I have one of those displays somewhere and will try out to figure where and how if at all.

openhardwarecoza, 2016-06-03:
Thats would help!  Thanks! (:

openhardwarecoza, 2016-06-12:
See https://plus.google.com/+PetervanderWalt/posts/gwesi8HBo9P added it to
LaserWeb3.  Thanks!

closed 2016-09-27 by openhardwarecoza