LaserWeb/deprecated-LaserWeb2 #68: Responsive/continuous jog for GRBL

SloMusti, 2016-06-03:
Based on https://github.com/grbl/grbl/issues/243#issuecomment-19505111 there is a way to have a responsive jog that has the movement while key pressed, would be super interesting.

A bit tricky to implement is the current version I believe as that would then be firmware specific as well and there would need to be a minimum time, so pressing the button moves for the specified distance, if pressed it moves until released. Could be implemented also as in Control tab with "Jog by continuous or so"

Relevant files:
https://github.com/openhardwarecoza/LaserWeb2/blob/a2145c5695e332be71c1374887c23eb99cf413a8/js/jog.js

Might be implemented through functions in here, depends on other comms modules.
https://github.com/openhardwarecoza/LaserWeb2/blob/0756443e480d3e795a89e7f3b0013d8b514ae344/js/grbl.js

Potential enhancement, here for reference as there is useful info on the link.

openhardwarecoza, 2016-06-03:
> Based on grbl/grbl#243 (comment)

I only accomodate cross platform features.  Must work on smoothie,
marlin,  and grbl

Cinezaster, 2016-06-03:
This could be handle by the comms layer eventually.

closed 2016-09-27 by openhardwarecoza