LaserWeb/deprecated-LaserWeb2 #72: Connect in menu bar missing

SloMusti, 2016-06-14:
Tested LaserWeb2 about 10 days ago just fine, now however there appear to be only Open and full-screen buttons in top bar when on Chrome+Win7, worked fine on the same machine before. Something broke I guess. New version looks awesome already, but would be nice to use this one meanwhile.

cojarbi, 2016-06-14:
@slomusti LW2 is just a CAM now. All efforts are on LW3 which is running fine for smoothie.

SloMusti, 2016-06-14:
Are there any live versions we could use with Grbl as we had people running machines that way.

cojarbi, 2016-06-14:
Maybe Look for an older version in github and run it locally.

cojarbi, 2016-06-14:
On windows cmd root do

git clone https://github.com/openhardwarecoza/LaserWeb2.git
go into laserweb2 directory
run " npm install "
run http-server to execute it

beside this you still need to run SPJS locally

SloMusti, 2016-06-14:
Made a for to my github and have it running there: https://github.com/IRNAS/LaserWeb2 for use until replaced by new version. Making the change of features clear in https://github.com/openhardwarecoza/LaserWeb2/pull/73 request.

This issue can be closed from my perspective.

cojarbi, 2016-06-14:
Good, thanks

closed 2016-06-14 by cojarbi