LaserWeb/deprecated-LaserWeb2 #75: Check this .dxf

r0773nluck, 2016-07-10:
when i try to generate this .dxf into gcode it takes all circles/ovals/arcs and places them at the origin. I am exporting the dxf out of fushion 360 and also tried and addon to convert polylines and had the same error. it does it with all arcs in dxf i generate in fushion 360, but no always to the origin sometimes just to the y0 location.

r0773nluck, 2016-07-10:
[dxf.zip](https://github.com/openhardwarecoza/LaserWeb2/files/355846/dxf.zip)

openhardwarecoza, 2016-07-10:
[image]
[image]

I dont see a problem when opening on my side? 

Show screenshot?

openhardwarecoza, 2016-07-10:
Elipses are ignored.  Convert that to Polylines...

openhardwarecoza, 2016-07-10:
Ok, managed to replicate...

[image]

Investigating cause now (suspect it started happening after the recent cnc-cam additions, was still working fine last week )

cojarbi, 2016-07-10:
I could replicate yesterday as seen bellow but i remembered i did a mini
wheel plate last week and should have had the issue then

r0773nluck, 2016-07-10:
i just noticed when you opened my file you did convert to gcode the issue only shows up after converting to gcode and selecting gcode view

openhardwarecoza, 2016-07-10:
Yeah figured that out eventually.  Thats why adding a screenshot when opening an issue helps so much - less guesswork (;

Anyway,  in LaserWeb2 (openhardwarecoza.github.io/LaserWeb2) it still works.  

Since the Preview looks correct, it means the DXFrender etc  is fine.  So the issue must be in the ->gcode part. 
Thats entirely done in threegcode.js.  Luckily with Github we have a full version history of all the changes to that file:  See https://github.com/openhardwarecoza/LaserWeb3/commits/master/public/js/threegcode.js

I made a LOT of changes to it to support the CNC cam, so likely one of them broke it.  Its now a timoues task to roll back and bisect these changes to find which broke it. 

It should take a day or two to hunt it down.  In the meantime, pass your DXF through draftsight/qcad/autocad and explode the arcs/circles to plain polylines.

r0773nluck, 2016-07-10:
Cool thank you so much ill try processing thorugh draftsight sorry for the late screenshot response

openhardwarecoza, 2016-07-10:
I figured out why it happens!

DXF circle geometry is drawn with two sets of coordinates:  Local (references line segments in relation to the center point of the circle) and "World" which relates to the rest of the drawing.  
In laserWeb2 the object used to create the Gcode is present in the THREE.js scene, thus when I call localToWorld() in threegcode.js,   since it is in the scene, the maths knows how to relate its local coordinates and compensate those to world coordinates.  

In Laserweb3 - we dont generate off the Scene, we generate from THREE.objects in an array.  Since this array isnt in the scene, the localToWorld() can't comnpute where they are in relation to the drawing.

Now to come up with a solution to this.  Not as easy. I recall I had a similar issue with the preview stage back in February... Got to figure out how to convert local to world for objects without using the dom.

r0773nluck, 2016-07-10:
i wish i could help with that but on a side note i imported my dxf into draftsight how do i convert all things to poly lines?

openhardwarecoza, 2016-07-10:
i shared a link in G+ this morning with you in that comment thread

openhardwarecoza, 2016-07-10:
Here we go:  See https://github.com/openhardwarecoza/LaserWeb3/commit/c8afc0b7a846ba12b4b9180413126974c65215aa

And now it looks like:

[image]

Test and let me know if this fixed your issue (:
If it did, consider buying me a beer (; http://openhardwarecoza.github.io/donate/

r0773nluck, 2016-07-12:
well you did it amazing

closed 2016-09-27 by openhardwarecoza